sn4k3/Ender3 #3: Compatible printer strings?

talldonkey, 2018-12-19:
Hi - Thank you for keeping this up to date.

I was curious about your latest commit so I took a look at the diff and noticed that there's some mentions of Prusa where I think creality and ender 3 was intended, see below examples.

`compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and printer_notes=~/.*PRINTER_HAS_VOLCANO.*/ and nozzle_diameter[0]==0.6`

When I was expecting to see Creality instead of Prusa.  The *theoretical* effect would that these profiles would not show up when the Ender-3 is selected (I haven't tested it out for myself)

`compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_CREALITY.*/ and printer_notes=~/.*PRINTER_MODEL_ENDER3.*/ and nozzle_diameter[0]==0.8 and printer_notes=~/.*PRINTER_HAS_VOLCANO.*/
`
I see this in mostly the new 0.6 nozzle files.

Was this intentional, (maybe I'm misunderstanding the reason why Prusa is in there)

sn4k3, 2018-12-19:
Hello,

The reason why i put that is because i have i have both MK3 and Ender 3 with same config (Direct Drive + Volcano + E3D) So to avoid edit two times the profiles i had put that to allow me share the profile on MK3 and Ender3, if you look at printer (Creality DD) you can find that:

Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.
PRINTER_VENDOR_CREALITY
PRINTER_MODEL_ENDER3
PRINTER_HAS_VOLCANO
PRINTER_HAS_DIRECT_DRIVE
PRINTER_VENDOR_PRUSA3D
PRINTER_MODEL_MK3
MAX_VOLUMETRIC_25
MAX_SPEED_85

So in end is just to easy my life, and yes if you have Prusa MK3 you can use that too.
To convert in bowden you just need to change Printer Tab config.

Also i can't support every possible config, so take one and modify at your will, my start gcode would not work very well for the commun pleople since i have tons of mods include linear rails, so never forget to test and adjust printer tab!! All settings i have are tested by me when i use printer in that configuration.

Regards

talldonkey, 2018-12-21:
Hi, Ah - now that make sense now.  Great.  I also happen to have a MK3 Also
so this repo would be even more useful now!

Another question - you have BLTouch, and BLTouch Recall.  What's the Recall
version for?  I don't have a Bltouch but an EZABL, and would need to modify
the start codes, but wondering what it is - googling for "BLTouch Recall"
doesn't help much

sn4k3, 2018-12-21:
Yes recall is a name i gave myself, mean it will not run mesh bed but load it from last run values stored on EEPROM instead, so it's Recall ^^ and saves time, but last values must be there and ok (bed same level).
It work will ABL too but i recommend to run every time since ABL is faster than bltouch to probe i think...