sn4k3/Ender3 #8: Intro priming line collides with paper clips

probonopd, 2019-02-03:
The intro priming line collides with where most people would normally put their paper clips.

Hence, the intro priming line should be moved e.g., to the side, where no paper clips are usually installed.

sn4k3, 2019-02-03:
The profiles provide a base for users, the start gcode must be tuned by the user itself, since Ender3 is heavy modded i can't provide every case-of-use profile. The prime line is based on stock Ender3 (without ultrabase). BTW i also have a custom base and i hate clips because of that and smaller area, so i use Kypton tape all arround +/- 5mm edge. Have hard objects under nozzle area is always a bad idea

sn4k3, 2019-02-04:
I can do that latter. But you can see my 0.6mm DD profile which have prime line that way and replace the gcode for your profile. You can also send a pull request with that mod

sn4k3, 2019-02-08:
Change prime to Y on new release

closed 2019-02-08 by sn4k3

probonopd, 2019-02-08:
Thanks. :+1: