sn4k3/Ender3 #10: Under extrusion on small cylinders with possible fix.

abrileroux, 2019-02-08:
Hi there,
First of all thank you very much for the profiles for the ender 3. I'm getting the best prints I've ever had with your profiles.

I ran into an issue where the whole print would work beautifully until I got to a small cylinder/chimney shape. The printer would under-extrude badly for that shape resulting in a failure. The same file would print fine when sliced with Cura. 

[https://imgur.com/a/LZ0PhBn](url), Pictures of problem.

The solution I found was to remove the following line from the start Gcode : 

`M221 S{if layer_height<0.075}100{else}95{endif}`

It seems that the flow rate of 95 was just low enough to fail.
A flow rate of 100 doesn't cause any problems with the print, so I haven't noticed a downside to removing it (yet).

Just wanted to share my finding. Thank you again!

yarus23, 2019-02-08:
95 is too low for quality supports that are printed with higher speed. So 100% is the best value

sn4k3, 2019-02-08:
95 was the best value i found after tune e-steps. Are your e-steps extruding 100mm out of 100mm?
My ender3 come with esteps a bit off so i tunned first and 100% at slice gave me overextrusion, that's why i set it to 95.
Please tell me about your e-steps value

abrileroux, 2019-02-08:
> Please tell me about your e-steps value

Hi there, sure. My E-steps is at 100.4.

Thanks

sn4k3, 2019-02-08:
Humm a bit higher then my was. My was 98 for get perfect 100mm out of 100mm PLA

sn4k3, 2019-02-08:
Ajustment removed

closed 2019-02-08 by sn4k3