sn4k3/Ender3 #13: Filament priming

sfjuocekr, 2019-06-13:
The feed rate during the priming strip is a little high.

The first part comes out okay, but during the second part the extruder gear slips and if your coupling like mine is a little wonkey it will slide out as it is the weakest link after the extruder gear.

Personally, I don't think overextruding with force is required to prime the nozzle.

sfjuocekr, 2019-06-13:
Also, the start gcode differs between the profiles. I think you forgot to update the others, for example stock still has the jerky zig zag!

sn4k3, 2019-06-13:
The printer tab must be configured to the user preference and needs.
What printer setting you are using?

sfjuocekr, 2019-06-18:
Ender-3

If you do a diff on the stock profile vs one of the recently changed ones and grep out the start gcode line you will notice the differences.

sn4k3, 2019-06-18:
You mean Ender3 Stock? I know stock are different from others in prime.
I don't have stock anymore to test and try, my bed is also different coordinates and size.
On installation change printer tab and tune to your preference is required and a step of instalation, i can't cover a all in one profile as ender3 is the most hacked printer. When i used the stock profile in the past my extruder never click or skip steps at prime.
But if you can fix it, test and send gcode back, i will update the default for stock and bltouch which have same prime

closed 2019-07-13 by sn4k3