sn4k3/Ender3 #14: this is not an issue.. just a question

phamre, 2019-07-10:
Hello.. 

i like your work.. thanks

i have direct drive setup with v6 hotend and bmg extruder.. 
u think i can use your volcano direct drive profile? 
im not sure what differance between v6 and volcano hotend

Thanks again

sn4k3, 2019-07-10:
Sure you can use. Just go printer tab and remove volcano from comments. OR reduce max speeds by half at all infills

closed 2019-07-13 by sn4k3

DAddYE, 2019-07-24:
What exact setup are you using? I’m same as OP but I’ve a 0.6 volcano nozzle. Can I use volcano profiles?

I’m also wondering if i change to a 0.4 nozzle should I change all values in the Extruder setting as well... right? 

Lastly, what’s the difference with bl touch and the one with recall?

Thanks mate!

NovaViper, 2019-07-24:
@DAddYE The recall one basically is the same like the regular bltouch one with the exception that it loads the mesh data that you set before (and not probe the bed before each print) while the regular bltouch profile *always* probe the bed before starting the print

sn4k3, 2019-07-24:
> What exact setup are you using? I’m same as OP but I’ve a 0.6 volcano nozzle. Can I use volcano profiles?

I'm using 0.6mm Volcano DD (Direct Drive)
You can and should use it, but if you have bowden you need to change two parameters only (retraction is all you need to change in fact)

> I’m also wondering if i change to a 0.4 nozzle should I change all values in the Extruder setting as well... right?
> 

If so you need to recreate profiles and printer, mostly you will only incrase infills speeds but only up to a safe margin which i set to be +/- 150mm/s

> Lastly, what’s the difference with bl touch and the one with recall?
> 
> Thanks mate!

Already anwsered

DAddYE, 2019-07-25:
Thanks mate! I got this:

MK8 Bowden Extruder BMG extruder + V6 HOTEND Dual Drive Extruder for 3D Printer High Performance for I3 3D Printer - (Size: BMG 24V All-Metal) https://www.amazon.com/dp/B07KVHG5FH/ref=cm_sw_r_cp_api_i_nOsoDbTB4YVFN

I’m on direct drive but I have a v6 hot end.

phamre, 2019-07-29:
> Sure you can use. Just go printer tab and remove volcano from comments. OR reduce max speeds by half at all infills

i dont understand what comments i need to remove.. and when i change Directdrive profile and save with a new name.. all printer profiles so are not there longer.. 

maby you can take a screenshot and show what i need to do?

sn4k3, 2019-07-29:
@phamre tell me your setup (noozle size, volcano or v6, Bowden or direct, bed probe)

DAddYE, 2019-07-30:
The volcano option, AFAIK, only enables specific filament profiles. 

Do the options in the notes do anything else?

sn4k3, 2019-07-30:
Notes are used to show/hide profiles and filaments based on volcano. Also used for linear pressure

phamre, 2019-07-30:
> @phamre tell me your setup (noozle size, volcano or v6, Bowden or direct, bed probe)

@sn4k3 My Setup: 0.4mm Noozle, v6 hotend(trianglelab) bmg extruder(trianglelab) Direct Drive, EZABL Probe

sn4k3, 2019-07-31:
ok do the following:

1) Duplicate Ender 3 stock + BLTouch
2) Go Notes and remove "PRINTER_HAS_BOWDEN"
3) Go extruder 1 and set Retraction Lenght to 0.8mm or your best value.
4) Done - Do print test

PS: I notice crazy accel values for stock due new slic3r version, will fix them and push them, please download new profiles from master

plazmax, 2019-07-31:
Ty sn4k3 i have simular trianglelab e3dv6, titan with pancake directdrive except abl i can do same with ender 3 stock , btw i installed prusa slicsr 2.0 too but slicer pe works better with ur setting cant wait new update thank you

sn4k3, 2019-07-31:
Master updated

phamre, 2019-07-31:
> ok do the following:
>
> 1. Duplicate Ender 3 stock + BLTouch
> 2. Go Notes and remove "PRINTER_HAS_BOWDEN"
> 3. Go extruder 1 and set Retraction Lenght to 0.8mm or your best value.
> 4. Done - Do print [...]

Thanks i will do a test now. but i forgot to say im using Linear Advance

sn4k3, 2019-07-31:
Ok then you need to define the K for each filament. Add a printer note like: PRINTER_FULLMODEL_ENDER3_DD_04
After go filament tab under "Custom G-Code" define your K and use your PRINTER_FULLMODEL_ENDER3_DD_04, example:

M900 K{if printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{elsif printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and nozzle_diameter[0]==0.6}15{elsif printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and nozzle_diameter[0]==0.4}30{elsif printer_notes=~/.*PRINTER_FULLMODEL_ENDER3_DD_06.*/}0.04{elsif printer_notes=~/.*PRINTER_FULLMODEL_ENDER3_DD_04.*/}0.04{else}0{endif} ; Filament gcode

phamre, 2019-08-03:
> Ok then you need to define the K for each filament. Add a printer note like: PRINTER_FULLMODEL_ENDER3_DD_04
> After go filament tab under "Custom G-Code" define your K and use your [...]

ok. im not sure i understand.. what i need to add to notes in printer tab and filament tab? for  my filament Kfactor is 0.15

sn4k3, 2019-08-03:
Not right... If you use only one printer you can ignore that all ifs and notes. That's only usefull when you use multiple printers. So you can set M900 K0.15 alone instead all that mess.

phamre, 2019-08-03:
> Not right... If you use only one printer you can ignore that all ifs and notes. That's only usefull when you use multiple printers. So you can set M900 K0.15 alone instead all that mess.

only M900 K0.15 under Filament Tab -> Custom G-code ?

sn4k3, 2019-08-03:
Exactly