sn4k3/Ender3 #16: Infilll and Solid Infill and Max speeds for 0.2 print profile (stock) is way too high for certain filaments

NovaViper, 2019-08-09:
Hi there! I wanted to report an issue relating with the Infill/Solid Infill and Max speeds for the stock 0.2 print normal profile. For some odd reason, they're set REALLY high (80 for the infills and 90 for the max speed to be extact) while every other profile is set between 50-71. For certain filaments (PETG in my case) this is *way* too fast for the infill to handle and it will immediately get ripped off the model and make basically large gashes in the model (which inventibly causes print failures). 
Currently, I've set those settings all to 56 and that seemed to have fix my troubles

sn4k3, 2019-08-09:
Hello, Speeds are based on MVS, for 0.2mm profile the 80mm/s seens ok for normal PETG (80*0,2*0,45
= 7,2mm^3/s) While PETG is cap at 8mm^3/s on filament tab, so is under the PETG limit.
Now it can depend on temperature and filament you using, can you post your filament brand/name and temperatures you using? Also the motor and gear can influence max MVS. 

Instead of mess with speed which affect all other filaments, you can clone PETG filament and set "Max Volumetric Speed" to a lower value, and speed will ajust according.

NovaViper, 2019-08-11:
I tried with ColorMe3D Blacker Black and Haze Purple, and Amazon Basics Purple. All running at 235C for the hotend and 75C for the bed

sn4k3, 2019-08-11:
That's strange, 56mm/s seens low for infills given that profile, i renember i had no problem with Prusa PET when i have stock, now i can't test anymore since no longet stock.
Do you have lastest master profile? Because i updated machine limits which were in crazy numbers

NovaViper, 2019-08-11:
Oh! So it seems I do have an older version of the master profile. I'm updating my settings now :)

sn4k3, 2019-08-11:
> Oh! So it seems I do have an older version of the master profile. I'm updating my settings now :)

Update it and report back

NovaViper, 2019-08-12:
I just got new profile settings added and I did decrease the volumetic speed to 7, though I still have some blobbing.. I'm going to go and switch out my nozzle (I have a 0.5 nozzle installed) and see if that's what is causing the issue

sn4k3, 2019-08-12:
Wait you have 0.5mm nozzle in but using the 0.4 profiles?

NovaViper, 2019-08-12:
Yeah, I had to restort to switching it because I kept having issues printing 0.32 layer heights with my 0.4 nozzle

sn4k3, 2019-08-12:
You can't use 0.4mm profiles with a 0.5mm nozzle, there are change that must be done, including on printer tab...

but 0.4mm nozzle can do 0.32mm layers, the problem you have maybe related to other thing (If speed down doesnt help)

NovaViper, 2019-08-13:
For some reason after that first layer, it started failing again once I went back to my 0.4 nozzle. The filament settings was all stock (except for my temps) .-.

NovaViper, 2019-08-13:
[image]

sn4k3, 2019-08-13:
Look likes more a height problem than speed. Does your extruder click?
Try put 5 at volumetric speed under filament tab

sn4k3, 2019-08-13:
Use 40x40 square tests to speed up tests

NovaViper, 2019-08-13:
I didn't hear the extruder clicking, and whenever it prints perimeters and walls, it doesn't have the issue shown. Im currently releveling the bed just in case it's off from my last print in PLA

NovaViper, 2019-08-13:
Ok.. so I got a 50x50x1.28 mm rectangle going.. and it seems to be much better at 5mm^3/s

sn4k3, 2019-08-13:
IF better but not perfect try lower, like 4

NovaViper, 2019-08-13:
Welp, it did this in the end 
[image]

NovaViper, 2019-08-13:
But all the other layers were fine (but I didn't catch the final or the one before it to thoroughly confirm it)

NovaViper, 2019-08-13:
I can confirm that it did indeed do that in the very top layer
[image]

NovaViper, 2019-08-13:
There was some issues in the layer right under it to.. it looks like at one particular spot it didn't stick properly
[image]

NovaViper, 2019-08-13:
*Sigh* The 4mm^3/s top layer had issues, no layer had issues 
[image]
[image]

sn4k3, 2019-08-13:
Well set MVS to 1 (lower than TPU xD) if you still have problem is not related to speed.

NovaViper, 2019-08-13:
Should I turn off the cooling fan entirely? The cooling fan seems to kick on at the very last layer

sn4k3, 2019-08-13:
> Should I turn off the cooling fan entirely? The cooling fan seems to kick on at the very last layer

You can try, but i guess not the fan causing the problem

NovaViper, 2019-08-13:
I just started the print at 1VMS and no fan for the PETG. I'll try the PLA one afterward

NovaViper, 2019-08-13:
This time at 1 MVS and the fan off,it came out *perfect*

sn4k3, 2019-08-13:
If MVS value is respected that mean +/- 7mm/s speed. Can you confirm that is that slow?
Now you can reenable fan as it was

NovaViper, 2019-08-13:
It came out *really* slow.. took an entire hour to finish the print instead of 15 minutes

NovaViper, 2019-08-13:
[image]

sn4k3, 2019-08-13:
Can you try now with PLA?

NovaViper, 2019-08-13:
Alrighty. im starting the PLA print now

NovaViper, 2019-08-13:
The PLA one came out perfectly too, I'm using all the stock settings in the Generic PLA profile

sn4k3, 2019-08-13:
With stock speeds right? 
Then something is wrong with filament, do you have other pet brand? Humidity can ruin a spool

NovaViper, 2019-08-14:
I do have other brands, but they all have had the same issue

sn4k3, 2019-08-14:
Well if problem presists on PET the best bet is test the max MVS your printer steup support. a good way to set that is print with stock profiles. At second layer when problem comes start slowing down the printer speed using the knob (FR) up you see a good line pattern. Then look at FR percentage, lets say it's 75%. Pick stock MVS and multiply by 0.75, eg: 8*0.75 = 6
Thats the way i pick the real max speeds, but again, max everything is not a general rule for everyone, a better extruder can extrude more, depend on each one setup

NovaViper, 2019-08-14:
Ya think I should also use completely stock filament settings from the Generic PET profile?

sn4k3, 2019-08-14:
To get your limit yes, and when found create or modify the profile with your value

NovaViper, 2019-08-14:
I just noticed that the infills are *really* high 
[image]

NovaViper, 2019-08-14:
It's not ripping up layers but it still looks a bit rough 
[image]

sn4k3, 2019-08-14:
> I just noticed that the infills are _really_ high with all stock settings

All settings print at same MVS on infills, speed is a illusion, printing at 40mm/s can be slower than a 20mm/s on other setup, what matter is the Volumetric speed (MVS). So all profiles are boosting infill and solid infill to max. capability. Top and bottom infill are always slower to get a better finish.

Find your best infill speed and use that as reference

NovaViper, 2019-08-14:
Just finished the stock settings print and it got better when I slowed it down to 70%,  but then for some reason the top infill got blobs in it

NovaViper, 2019-08-14:
[image]

sn4k3, 2019-08-14:
> Just finished the stock settings print and it got better when I slowed it down to 70%, but then for some reason the top infill got blobs in it

Try 60%, then 50% keep testing up to get the perfect look

sn4k3, 2019-08-14:
What mods you have on your printer? Complete setup? As i see is not stock stock

NovaViper, 2019-08-14:
The bed is just a plain glass pane I got from Home Depot coated with sugar water, the hotend is actually still the same, except I have mounted my extruder to be directly ontop of the hotend (hence it's direct drive now) and the fan shroud is the Petsfang Bullseye

sn4k3, 2019-08-14:
Ok then you should calibrate your extruder.
Mark 20cm with a pen on filament, using terminal or LCD extrude 100mm. When it stop check if mark hit at 10cm on the ruller. If no you need calibrate your e-steps.
Also as you have direct drive how much retraction you set on extruder?

NovaViper, 2019-08-14:
I used to have 1.8mm and 70mm/s for the retraction; when doing the tests, I started using 0.8mm and 35mm/s

NovaViper, 2019-08-14:
Also how do I calibrate e steps on a direct drive extruder?

sn4k3, 2019-08-14:
> I used to have 1.8mm and 70mm/s for the retraction; when doing the tests, I started using 0.8mm and 35mm/s

Thats seens ok as start point.

> Also how do I calibrate e steps on a direct drive extruder?

With the method i said before, marking, extruding and verify extruded lenght

NovaViper, 2019-08-15:
The e steps are fine, I marked 120 and told to extrude 100, and 100mm got extruded

sn4k3, 2019-08-15:
Ok, one more check: with steppers enabled (Holding) try to pull fillament by hand to see if gear rotate, it shall not rotate. If filament gets out mean gear grub is lose

NovaViper, 2019-08-15:
I can't get it to budge at all

sn4k3, 2019-08-15:
> I can't get it to budge at all

Well i'm out of possibilities, i guess you have to decrease speed for PET go with it.
As last check, go slicer and see if nozzle size is correctly set, as extrusions width on advanced (should be 0.45mm)

NovaViper, 2019-08-15:
Welp! I just I may have discovered what our culprit is! So the weather in my area has been going nuts lately (with it being abnormally hot and because it's Louisiana, the humidity is quite high) and my main air conditioning unit breaking due to the extreme heat, the humidity in the air must have gotten high enough to actually affect the prints. So I went today and let my window air unit run for the majority of the day. I just ran the print and the layers came out perfectly

sn4k3, 2019-08-15:
Humidity ruin pet spools, if you have that issue better build a spools dehumidifier. Or cook it at 50°c for day

NovaViper, 2019-08-15:
Ok I think my PETG doesn't like the new temp settings, mid way in printing out this case, I notice this MASSIVE clump of filament stuck on the nozzle and it literally jacked the entire print up

NovaViper, 2019-08-15:
[image]

Zviki68, 2019-08-25:
What's your nozzle temperature?

NovaViper, 2019-08-25:
> What's your nozzle temperature? And what slicer you are using?

240 and Prusa Slicer, I'm using all of the stock filament profile settings that's for the Generic PETG filament

NovaViper, 2020-01-11:
Hey im going to close this! I switched to a TriangleLabs V6 and my printer's being doing really well with the infills and PETG! Thanks for making the profile!

[3 more replies]

closed 2020-01-11 by NovaViper