sn4k3/Ender3 #17: Slicing with these profiles is very slow!

marksu78, 2019-09-04:
Slicing with these profiles takes a long time! PrusaSlicer 2.0.0 does the basic slicing quickly but then the "Generating G-Code" can take minutes.

When I slice the same STL with built in Prusa profiles it takes just few seconds!

As an example:
https://www.thingiverse.com/thing:3576952

To slice this with the unmodified settings of:
**Printer**: Creality Ender-3 Stock
**Filament**: Generic PLA
**Print settings**: 0,20mm NORMAL Ender3

The initial slicing takes **7 seconds**, Generating G-Code takes **3min 53 seconds**

Now if I change the settings to:
**Printer**: Original Prusa i3 MK3S
**Filament**: Prusament PLA
**Print settings**: 0.2mm QUALITY MK3

The whole slicing and G-Code generation is done in **10 seconds!**

A huge difference! What might be the reason why the slicing takes so much longer time with these profiles?

sn4k3, 2019-09-04:
Hello, disable avoid crossing perimeters. It's what causing the delay. The bigger the model the more it will take with that option

marksu78, 2019-09-04:
Yes, that helped thank you! I need to do some experimenting how much that feature actually helps, so if it's worth the extra time on bigger models

closed 2019-09-04 by marksu78

sn4k3, 2019-09-04:
That's only usefull for Bowden to avoid cross perimeters and then retracts, that's it. If your retracts make no mess or trouble you can put that off

marksu78, 2019-09-04:
Yes, I have now disabled it from all the print settings. I will enable it again if prints get messy with certain materials and then see if it helps.