sn4k3/Ender3 #22: Filament yanked off bed when travelling on first layer

diabl0w, 2019-11-04:
First of all, thank you for these profiles! They are amazing, except this one major problem I am having:

I am having an issue with the ender 3 0.24 FAST print profile where after the initial line that is printed across the bed in front from left to right, as soon as it finishes at the right side and goes to travel toward the center to begin the actual print, it yanks/pulls/curls up the last 10mm or so of filament off the bed with it. This also happens for basically any movement on the first layer of the print where it prints a layer and then stops to travel. Then the loose ends sticking in the air get bumped into and everything is a mess.

Can you recommend which settings I can add/change to fix this?

sn4k3, 2019-11-04:
Can you post a video?
What mods you have on your printer?
Do you change any settings on slicer?

diabl0w, 2019-11-04:
Solved by installing the latest version of slicer... I apologize, I should have tested that before opening an issue. Thanks for the reply!

closed 2019-11-04 by diabl0w