sn4k3/Ender3 #23: Compatibility/Implementation with New Ender Profile in 2.2.0 alpha

Astrofreak85, 2020-02-18:
Your profiles are good, the new Ender3 profiles in 2.2.0 alpha are a bit slower than yours and they don't have as many profiles as you.
You may ask if they can implement your profiles out of the box?

sn4k3, 2020-02-18:
They are still in alpha, let it get stable first. They will priority quality at a usable speed. My profiles push the machine a little bit and offer thicker extrusion selection to boost print speed. They support some nozzles but not volcano so far even for mk3. You can always use my profiles in pair with prusa, is not a problem, i guess they will not collide with names

closed 2020-02-18 by sn4k3

NovaViper, 2020-02-18:
> Your profiles are good, the new Ender3 profiles in 2.2.0 alpha are a bit slower than yours and they don't have as many profiles as you.
> You may ask if they can implement your profiles out of the box?

The profile as it is now works with the alpha 2.2.0