sn4k3/Ender3 #24: Invalid and Unknown commands in klipper

seebaer1976, 2020-02-20:
Hi
I use "Klipper" as firmware because I have a few invalid and unknown commands in klipper.

Invalid M204 command "N5 M204 T1250*82"
Unknown command:"M900"
Unknown command:"M280"

One parameter seems to be missing from the M204.
M900 and M280 are not supported by Klipper.

M900 is used differently on Klipper.
https://github.com/KevinOConnor/klipper/issues/92

NovaViper, 2020-02-20:
@seebaer1976 The profiles were made for Marlin firmware, not Klipper. You should be able to change the gcode to work with Klipper

sn4k3, 2020-02-20:
Ignore that errors, it will continue to print. M900 and M204 are configured in klipper config file and will not hurt leave them on gcode since they will no nothing.
As for M280 is bltouch related, it may have other command to klipper that you need to look for.

seebaer1976, 2020-02-20:
After reading back and forth for a long time and your help, I was able to solve it.

@sn4k3 sn4k3 M900 and M204 = pressure_advance, you had it in your config but I didn't have it in mine. commented out the two commands in the PrusaSlicer.
I could also solve M280, simply create a macro in the Klipper config:

[gcode_macro M280 P0 S160]
gcode:
BLTOUCH_DEBUG COMMAND = reset

Thank you for your help.

sn4k3, 2020-02-20:
Glad you fix it. Regards

closed 2020-02-20 by sn4k3

seebaer1976, 2020-02-20:
then it can be closed

darkstudio, 2022-09-03:
> After reading back and forth for a long time and your help, I was able to solve it.
>
> @sn4k3 sn4k3 M900 and M204 = pressure_advance, you had it in your config but I didn't have it in mine. commented [...]

Hi, when we create a macro with "SPACE" , Klipper refuse to restart firmware and show a error message =>

_**Name of section 'gcode_macro M280 P0 S160' contains illegal whitespace

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted**_

Space is not supported