sn4k3/Ender3 #25: Problems encountered with your config

superm1, 2020-03-11:
Hi,

I've recently tried switching to Prusa using your set of configs from Cura.  Overall I really love them, but I noticed two things I wanted to mention.

1. I don't seem to get good adhesion by default on my glass bed, even with your temperatures set.  I bumped up "disable fan for the first" to 4 layers, and this seems to help me.  Wondering why you picked one by default?  (Prusa default seems to be 3).

2. At the end of a tall but wide print I found the head came crashing down on the print when it reset since it resets to left and then back down to 0 Z height.  Cura's default was resetting to a higher Z and this never happened.

I compared cura's code and yours and it seems that for Cura they just reset to raise Z:
```
G1 Z10 ;Raise Z more
```

With yours I see it's an if statement.  Since I've been seeing it reset to effectively zero every time for me I have to wonder if one of those variables isn't working right as end G code.
```
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up
```

I double checked my octoprint log and this is what it sent:
```
G1 Z77.6*66
```

superm1, 2020-03-11:
Full set of last commands:
```
Send: N39210 G90*25
Recv: ok
Send: N39211 G1 Z77.6*66
Recv:  T:209.92 /0.00 B:60.36 /0.00 @:83 B@:0
Recv: ok
Send: N39212 G28 X0*80
Recv:  T:209.88 /0.00 B:60.19 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:0.00 Y:137.30 Z:77.51 E:24358.85 Count X:0 Y:10984 Z:31006
Recv: ok
Send: N39213 G1 Y180*98
Recv: ok
Send: N39214 M84*18
```

superm1, 2020-03-11:
Hmm, experimenting some it seems that sending the G1 Y180 is actually what's doing it which makes no sense..

sn4k3, 2020-03-12:
Hello, 
Glass beds are not the best in adhesion without additives, i never used glass and i rarely have adhesion problems. Profiles are also based on stock which have a very strong adhesion. 
You also have to make sure the bed temperature is stable and there are no air drafs (AC, doors open, air moving) Some brands of PLA are better than others on adhesion, you can try incrase 5ºc to bed temperature if having problems.

About reset, what mods do you have on printer? Any PowerResume or backup mod?

superm1, 2020-03-12:
Thanks for responding.

>Glass beds are not the best in adhesion without additives, i never used glass and i rarely have adhesion problems. Profiles are also based on stock which have a very strong adhesion.
You also have to make sure the bed temperature is stable and there are no air drafs (AC, doors open, air moving) Some brands of PLA are better than others on adhesion, you can try increase 5ºc-10ºc to bed temperature if having problems.

I've tried hatchbox and Amazon basics brands PLA and encountered this with both even after raising the temperature some.  The only thing that has helped so far for me was increasing number of layers before fan turned on. I did previously have a magnetic bed, but it got trashed from bending while removing prints and/or printing PETG on it.  The glass bed I have does have a black coating on the top though.

>About reset, what mods do you have on printer? Any PowerResume or backup mod?
Using last version of PrusaSlicer?

Only mods have been BLTouch and newer springs.  Otherwise stock ender-3 pro.  I used the BLTouch FW from creality's website, haven't yet tried to move to a newer version of merlin or anything like that.
Using Prusa 2.2.0-rc2.

I have reproduced it just by resending that set of reset commands I captured from octoprint's terminal too now, so I think that at least rules out prusa causing it.

sn4k3, 2020-03-12:
`I've tried hatchbox and Amazon basics brands PLA and encountered this with both even after raising the temperature some. The only thing that has helped so far for me was increasing number of layers before fan turned on.`

If the increased layers without fan works well for you then there are no bad to keep that settings. Pleople often use a layer of glue to glass to prevent warping, cura seens to squish more one first layer than slicer.

`I have reproduced it just by resending that set of reset commands I captured from octoprint's terminal too now, so I think that at least rules out prusa causing it.`

I'm always at stable releases, can you try with the stable PE? Anyway i will try that latter on my E3. Just print something and cut power then turn on again to reproduce the problem?

superm1, 2020-03-13:
>If the increased layers without fan works well for you then there are no bad to keep that settings. It can only impact very small models. Pleople often use a layer of glue to glass to prevent warping, cura seens to squish more one first layer than slicer.

Yeah I tried to print a tree model and some of the leaves started to warp; I think you're right "small models" still don't adhere well to the glass.
  
>I'm always at stable releases, can you try with the stable PE? Anyway i will try that latter on my E3. Just print something and cut power then turn on again to reproduce the problem?

Yeah I just finished a print done with prusa and your config, cut the printers power and then from Octoprint sent the following:
```
G90
G1 Z77.6
G28 X0
G1 Y180

```

On the G1 Y180 the Z axis reset too.

sn4k3, 2020-03-13:
>Yeah I tried to print a tree model and some of the leaves started to warp; I think you're right "small models" still don't adhere well to the glass.

Put a layer of glue stick, it will last many prints and easy to clean/mantain, that way is trouble free

> Yeah I just finished a print done with prusa and your config, cut the printers power and then from Octoprint sent the following:

I tried to reproduce the problem using your steps, my print does exactly what gcodes does and Z stay on same level untouched. I'm using last marlin bugfix but in all my versions it never happen to me, in fact on a power reset the end gcode always raized Z not lowered. If you sending G1 Yxxx it should move Y not Z, why printer thinks your Y is Z? Do you double check your configurations and wiring?

superm1, 2020-03-13:
> I tried to reproduce the problem using your steps, my print does exactly what gcodes does and Z stay on same level untouched. 

Yeah that's the exact behavior that I expected too..

> I'm using last marlin bugfix but in all my versions it never happen to me, in fact on a power reset the end gcode always raized Z not lowered. 

When you flashed last marlin bugfix release, any modifications needed?  Just put a bootloader on and then load new FW from Octoprint?  Sorry this part is new to me, but I think I can easily put a bootloader in place using the ISP that came with BLTouch.

>If you sending G1 Yxxx it should move Y not Z, why printer thinks your Y is Z? Do you double check your configurations and wiring?

I didn't touch the wiring from what was factory other than adding BLtouch.
Normally Y and Z commands work.  If Y and Z were swapped I would sure be getting some funky prints at least :)
So I have to think there is a bug in the FW build loaded on the printer related to this set of circumstances where the heads are or something.  Your commands seem sound to me.

sn4k3, 2020-03-13:
> When you flashed last marlin bugfix release, any modifications needed? Just put a bootloader on and then load new FW from Octoprint? Sorry this part is new to me, but I think I can easily put a bootloader in place using the ISP that came with BLTouch.

Very modifications are needed, or you can just put the stock config file from pre-configurations that marlin offer to your printer model and go that way, but still need to configure bltouch.
I recommend you to go line by line, read and understand what it does and activate what you need/want under configuration.h and configuration_adv.h
It's something to take your time, go slow and make sure it's all right.
About bootloader first try uploading a firmware, i think new models already have bootloader in.

superm1, 2020-03-13:
OK thanks for all your advice :)

closed 2020-03-13 by superm1

sn4k3, 2020-03-13:
You can find your machine configurations here: https://github.com/MarlinFirmware/Marlin/tree/2.0.x/config

Download Marlin
Put your machine configurations there
Start look and editing

shnee, 2020-03-24:
I am also seeing this behavior.

This part of the end code:
```
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif}
```
gets evaluated to
```
G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif}
```
in my gcode.

sn4k3, 2020-03-24:
> I am also seeing this behavior.
> 
> This part of the end code:
> 
> ```
> {if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif}
> ```
> 
> gets evaluated to
> 
> ```
> G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif}
> ```
> 
> in my gcode.

Update for the latest PrusaSlicer v2.2 and update profiles from last master

shnee, 2020-03-24:
You are correct, updating PrusaSlicer fixed this issue.

Thank you for your quick response and your work here. Your profiles have significantly improved the quality of my prints.