sn4k3/Ender3 #26: Better profile.

burnham, 2020-03-29:
Hi my friend,

First of all I want to congratulate you on your excellent work, it has been super useful.

Ok, I just want to make a question.

I have an Ender-3 printer with:

- [Metal V6 Hotend Bowen](https://es.aliexpress.com/item/32844028127.html?spm=a2g0s.9042311.0.0.274263c0qk4atK) with **Nozzle 0,4**

- [Bowden Extruder BMG](https://www.aliexpress.com/item/32917029058.html?spm=a2g0s.9042311.0.0.274263c0Rg10Wo)

- [3D TOUCH](https://www.aliexpress.com/item/32832887426.html?spm=a2g0s.9042311.0.0.274263c0Rg10Wo)

- [Printer board MKS Gen L](https://www.aliexpress.com/item/32802151924.html?spm=a2g0s.9042311.0.0.274263c0Rg10Wo)

- [MKS TMC2208 2208 Stepper Motor Driver](https://www.aliexpress.com/item/32888980385.html?spm=a2g0s.9042311.0.0.274263c0Rg10Wo)

Which profile should I use?

At the moment I use this configuration, but I have so many strings.

Thanks in advance.

[image]

sn4k3, 2020-03-29:
Hello, you are absolutely using the wrongest profile. You must use Creality Ender-3 Stock + BLTouch for yor setup.

burnham, 2020-03-30:
Thank you for you quick answer, you said "Note that title says 0.6 Nozzle, and you are using 0.4 Nozzle", in the picture says "0.6" but I just change 0.6 to 0.4 in the configuration in the profile I was using.
I will use  Creality Ender-3 Stock + BLTouch for my setup, but my question is, have this configuration any thing to do when I also use Bowden Extruder BMG?

Thanks.

burnham, 2020-03-30:
Ok, just try with your tips "Creality Ender-3 Stock + BLTouch" and I wanted to make a [retraction test](https://www.thingiverse.com/thing:2766430) but my printer start to turn crazy, it starts ejecting filament at high speed.

sn4k3, 2020-03-30:
> Ok, just try with your tips "Creality Ender-3 Stock + BLTouch" and I wanted to make a [retraction test](https://www.thingiverse.com/thing:2766430) but my printer start to turn crazy, it starts ejecting filament at high speed.

Did, you tune your e-steps when change the extruder? If you manual extrude 100mm it does really output 100mm?
You also have changed the firmware right, marlin?

burnham, 2020-03-31:
Asking your question I can print normally when I use CURA, and also I have controlled those 100 mm, I have changed the firmware too and I use the last version of PrusaSlicer 2.2, I did not really have problems with your other profiles except the many strings. I will try again and I will let you know.

Any way I would like to be in contact with you and let you know my experiences.

sn4k3, 2020-03-31:
I have reviewd the profile and i can't find anything wrong. Do you have LA enabled on firmware? If yes try disable it or you must find and tune the value for each filament

burnham, 2020-04-01:
What is the meaning of LA?

sn4k3, 2020-04-01:
> What is the meaning of LA?

Linear Advance --> https://marlinfw.org/docs/features/lin_advance.html

burnham, 2020-04-01:
I think that I know whats is happening.
I just try to print again this [retraction test](https://www.thingiverse.com/thing:2766430) with this configuration
[image]
As you can see everything seems to be correct in the configuration, then I try to print with the next result (I can not add video in github)
[Video with the process](https://anonfile.com/taa1hcl6oe/20200401_205031_mp4)

After that I did it correctly with CURA, the I start to make a research and I made a clean PrusaSlicer installation with your profiles and I noticed that "Creality Ender-3 Stock + BLTouch" has not PLA profile, I think is where the problem come as you can see in the next pic:

[image]

Thanks for your support.

sn4k3, 2020-04-01:
Yes i remove them because they are exactly same as prusa so as now new version include them i have removed to prevent duplicates. Click where it say Add/retirar filamentos and search and tick all filaments you want to appear

sn4k3, 2020-04-01:
I just updated the profile, try with it test and reply back

burnham, 2020-04-02:
> 
> 
> I just updated the profile, try with it test and reply back

Sorry, I was to busy with the family, I will test it tomorrow in the morning and I will let you know.

burnham, 2020-04-02:
I just did what you recommended me, the results are pretty good.
[image]
 I still have some strings but seems to be good.
[image]

Really nice.

sn4k3, 2020-04-02:
Ok, don't forget, different colors and filament types will have different results, it's impossible to have a fixed retraction value for every case. Also the result you are getting is pretty good. Stringing will always be present

closed 2020-04-02 by sn4k3