sn4k3/Ender3 #27: Quick question about which profile to use

NovaViper, 2020-03-30:
Hi! I just recently started using a 0.8 nozzle on my Ender 3. I'm wondering which profiles I should use for my current setup:

- I have a TriangleLabs V6 Non-Volcano with an Aimsoar BMG clone equipped in a direct drive setup, using [this particular mount](https://www.thingiverse.com/thing:3721746). I also have a pancake motor and guineue BLTouch too

Any recommendations on which profile to choose and how to set it up?

sn4k3, 2020-03-30:
Hi, 0.8mm nozzle without volcano don't bring any benefits other than vase mode with big heights. A 0.6mm nozzle with V6 give you more detail window at same finish speed. With 0.8mm get ready for slow print moves. As you can see i don't have a profile for that setup just because is not worth, but if you want i can build one. Even 0.6mm nozzle can hit volcano limit if you use my boosted profile.

None of the profiles will work without proper configuration first, do you want i made that bundle?

NovaViper, 2020-03-30:
With a 0.6 nozzle, can I achieve the same layer heights/reduced print times? I've got a couple of parts that will take at least 2 and half hours to print with a 0.4 nozzle, but with a 0.8 it cut down the time by 1 and a half hour merely because of the massive layer heights. Hence is why I opted for a 0.8 nozzle instead.

sn4k3, 2020-03-30:
> With a 0.6 nozzle, can I achieve the same layer heights/reduced print times? I've got a couple of parts that will take at least 2 and half hours to print with a 0.4 nozzle, but with a 0.8 it cut [...]

Not same layer heights, only up to 0.45mm height. 

Pratical example:
0.45mm layer with 0.6mm nozzle your max speed must be lower than 35mm/s == 10mm^3/s
0.45mm layer with 0.8mm nozzle your max speed must be lower than 27mm/s == 10mm^3/s

So as you can see 0.6mm nozzle can finish the print at same time as 0.8mm nozzle by tunning speed.
Even with 0.4mm i can boost your extrusion and spare a ton of print time.
Please consider volcano, it worth even for 0.4mm nozzle, but for larger sizes are a must.

Now if you don't need the extra detail from a 0.6mm nozzle, then theres no bad in use a 0.8mm, the down side is you will never able to print lines lower than 0.8mm width.

0.4mm nozzle window (0.4mm-0.7mm)
0.6mm nozzle window (0.6mm-1.2mm)
0.8mm nozzle window (0.8mm-1.7mm)

I have to go sleep now, but tell me if you want keep with 0.8mm and i'll make a configuration tomorrow so you can test, and yes it will give you a speed boost over 0.4mm nozzle

sn4k3, 2020-03-30:
Before i go i just released a pre setup that you can use. Get last master and test with the new printer profile.
Get ready to see your printer very slow moving, but actual printing faster

NovaViper, 2020-03-30:
Cool, thansk! I don't mind switching to the 0.6 nozzle and yeah, I do notice it moves considerably much slower and I lost a bit of accuracy since switching to the 0.8 nozzle. Ok so, if I go with the 0.6 nozzle, what's the maximum possible speeds I can go with, for say with PETG?

sn4k3, 2020-03-30:
Speed (mm/s) doesn't matter,  Volumetric flow is what count and messured as mm^3/s. For v6 and PETG i recommend a maximum of 8mm^3/s.
You can get the volumetric speed with that formula: Layer Height * Extrusion Width * Speed

So, example: 
0.30mm layer * 0.65mm extrusion * 41mm/s = 8mm^3/s --> So your max speed for that setting combo should be 41mm/s

NovaViper, 2020-03-30:
Ah ok. So in that case, i'll switch over to the 0.6 nozzle. Er, do you have a profile you recommend me starting off from?

sn4k3, 2020-03-30:
I don't have any, but good news, you can use the Original Prusa MK3 0.6mm nozzle.
You just need to copy printer and change bed size, height and introduce bltouch code.
I can also publish a bundle

NovaViper, 2020-03-30:
Ah cool! Thanks!

NovaViper, 2020-03-30:
Oh question, so would I use the MK3 profile from Prusa or the one that you have in your bundle?

sn4k3, 2020-03-30:
Any will do, i just uploaded your config, please update

NovaViper, 2020-03-31:
Hey uh, the 0.6 profile that you made seemed to have doubled the print time. It used to take about 12 minutes to print a 20x20 hollow calibration cube when using the 0.4mm nozzle @ 0.32 layer height, now it takes 30 minutes using the 0.45 layer height

NovaViper, 2020-03-31:
*Oof*! I just realized I was looking at my print estimator on Octoprint (and it was WAY off). The print time seems to be right about the same

sn4k3, 2020-03-31:
> _Oof_! I just realized I was looking at my print estimator on Octoprint (and it was WAY off). The print time seems to be right about the same

Always compare with PrusaSlicer estimator.
The 0.6mm noozle have 2x fastter results compared with 0.4mm at 0.32mm.
The problem is you printing a small object (20x20), auto speed and cooling will come in play and slow down your print, so thats normal to be the same. You also have the small perimeters which will cut speed so keep in mind that 20x20 is not a good test, you will only notice the diference on real prints

NovaViper, 2020-04-06:
Hey, I've been trying to setup the 0.6 profile you made but I noticed there are several profiles missing, like, there's no 0.2 or anything below 0.3. And there's a couple of materials missing too (TPU and PLA specifically)

sn4k3, 2020-04-06:
> Hey, I've been trying to setup the 0.6 profile you made but I noticed there are several profiles missing, like, there's no 0.2 or anything below 0.3. And there's a couple of materials missing too (TPU and PLA specifically)

I have added now the missing profiles.
About filaments read https://github.com/sn4k3/Ender3/issues/26#issuecomment-607474282 OR readme point 2)

roomme13, 2020-04-21:
Hi, BMG extruder in remote setup not called as and direct drive right? It should same as BMG as directdrive but with small tube to the heatsink? Or it's different?

Rollback from direct volcano to stock ender3 with remote BMG, as every time I  print  the volcano clogged.  The funny thing is when print starts the nozzle clogged, when I stop and try  extrude manual, it work normally, even the first stroke before print is in good visual. Frustrating about this and roll back to stock ender hotend.

sn4k3, 2020-04-21:
Right, what ever you use in remote setup is always bowden. Direct Drive is where you have your extruder gears just close to hotend as possible.

The clog might be another problem, you need troubleshoot first before give up. 
Is V6 heatsink + volcano?

roomme13, 2020-04-21:
> Right, what ever you use in remote setup is always bowden. Direct Drive is where you have your extruder gears just close to hotend as possible.
> 
> The clog might be another problem, you need troubleshoot first before give up. 
> Is V6 heatsink + volcano?

So stock printer profile should be fine if i use bmg as remote drive?

I think so, at the first I thought it retraction problem, but even I make it 0.5mm, still clogged.

Yes it's V6 heatsink with Volcano, 4020 fan heatsink.
The clogged always happen in heatblock part of heartbreak.

sn4k3, 2020-04-21:
Yes you should use the stock profile if using 0.4mm nozzle and bowden

So i would strict follow e3d instalation guide, heatbreak must be tight agaisnt nozzle at correct height. If extrude by hand works fine, then it can be antoher thing, like LA, if you have it enable on marlin, just disable it

gminadak, 2020-05-09:
@sn4k3 Thank you for the work and effort you have make to put all this together.
I'm using your profiles with great success for almost a year.
If you can and have the time please post some info from your setup with pictures, i will upgrade from stock and i thing the best option is to go with volcano and BMG (now i have a Titan clone direct with stock hotend).
But i don't know what parts to choose e.g. For the volcano "all metal" or ptfe and the extruder BMG or the mini BMG and what parts fan to use with this for direct setup.
Thank you for your time sorry if i post in the wrong section but i did't want to open a new thread.

sn4k3, 2020-05-09:
Hello, at this point i don't recomend BMG,  go Hemera instead as BMG still have a long path from guears to extruder. I use skelestruder at my ender 3 because of that, very short path. But now you have Hemera to go with... BMG is same price if you go orginal, of course clones are cheap. So if you have the money don't look back and pick hemera.
About block always go All Metal! PTFE will disallow higher temperatures such for ABS and higher, doing so can kill you in a mid/long time.
My recommendation: Hemera + Platted Copper Volcano + Platted Copper Nozzle + Titanium Heatbreak

gminadak, 2020-05-09:
I'm using only cloned parts on my printer ;).
Hemera looks the perfect combination to go this days but its way above the money i'm willing to spend for my ender.
The Skelestruder looks like beast! and i like it, i will search for more information on this setup.
Thank you for the quick reply.

sn4k3, 2020-05-09:
I dont know if they already cloned hemera, maybe yes but under other name? Well you can always go with what you already have. BMG make sure to use a design that clamps heatsink directly to avoid a longer path.
Skelestruder works very well, and while almost part free you need to print many parts, not easy for start but it get better as you start assemble. I recommend you have another printer just in case of any part failure

gminadak, 2020-05-09:
The reason i'm considering to upgrade to volcano is to solve the problem with this lines, i have try everything but even with the "direct drive" the problem is there.

[image]
[image]

sn4k3, 2020-05-09:
That doesn't seen a extruder problem... Part of your problem may be  the poor Z.
Best thing you can do is upgrade to dual Z and give it proprer care.
Also check your belts, check your wheels nuts every wheel must rotate and grab the frame. With your hand try to stop a wheel while its moving, it is hard to stop its ok, otherwise if you easy stop then just with a touch mean they need a bit more tight

sn4k3, 2020-05-09:
Here a mod for super strong dual Z: https://www.thingiverse.com/thing:3612465

gminadak, 2020-05-09:
The Z was the first thing i had in mind so i printed this test from dpetsel and the problem is almost gone.
The big part is with seam aligned on back the other in VASE mode.

[image]
[image]
[image]
[image]

sn4k3, 2020-05-09:
The lines are still there, some are more squish then others, trust me tune your Z is the weak point of ender3

gminadak, 2020-05-09:
Thank you for your help i will recheck my Z and wheels

sn4k3, 2020-05-09:
for example, move your Z just 1m up other down other up other down, and check if right side move or stay still.

gminadak, 2020-05-10:
I did check my wheels and X the X is aligned with the top and my Z extrusions have the same length.
What i did and i see some improvement was to hand tighten the lead screw nut bolts and then turn back approximately one turn.
The problem is still there but less noticeable.
I have observe the right side of the X gantry and i see the movement by hand and when the printer running with the Z hop. 
I had over tighten the Wheels on this side and it was like losing some movement now seems better and that reflects to prints but still not perfect i can also move the right side by hand a little.
I think i need to add a second Z gladly i already have the motor from the stock extruder.
Your dual Z setup looks nice but what filles i need to print and do i need MGN linear rails for this mod?
Thank you.

sn4k3, 2020-05-11:
Lack of dual Z is the worst problem on all enders

gminadak, 2020-06-13:
Hello @sn4k3 how are you?
So after many prints and playing with he settings
I have discover a new problem when i try to print something cylindrical there is no way to complete hide the seam i have try random, aligned, nearest but nothing works is this the way it should be?

Below is a picture of the prints and my configs if you want to take a look 
[image]

[slicer.zip](https://github.com/sn4k3/Ender3/files/4774833/slicer.zip)

Thank you.

sn4k3, 2020-06-13:
Hi, see this anwser: https://forum.prusaprinters.org/forum/prusaslicer/prusa-slicer-2-0-how-would-i-go-about-to-get-rid-of-that-vertical-seam/

Your best bet is tune your printer and if in use of LA get the best K factor.
Cylindrical shapes are  the worst

gminadak, 2020-06-13:
I did check my e steps again and it was off by 1.4mm looks better now but still visible.
Thank you for your help.

closed 2020-06-13 by sn4k3