sn4k3/Ender3 #29: Profiles not loaded by slic3r

Venice-89, 2020-04-03:
Hi,

I am using your Slic3r for a while. Many thanks for your great work !

Now I made an updates of your Profiles on my existing slic3r 2.1 but your profiles doesn't appear in the drop down menue, only the printer profiles, not the print or filament profiles.

I tried to delet everything and started from scratch but no luck.
Then I updated to Slic3r 2.2 and but with the same problem :/

Do you know what the problem could be?

sn4k3, 2020-04-03:
Hi, do you strict follow the instalation guide from readme at project page?
Also for filaments  please read this: https://github.com/sn4k3/Ender3/issues/26#issuecomment-607474282

Venice-89, 2020-04-03:
Hi,

yes absolute, I followed the instruction but only the printers are added to the dopdown menue.
But it's possible to load the profile via import mhhh

I will try it on other notebook again

sn4k3, 2020-04-03:
Can you attach a printscreen of the slicer and dropdowns?

Darisson1, 2020-04-11:
Unfortunately I have the same Problem. I will reinstall everything tomorrow and I will make Screenshots. 
Maybe you can help me and we can solve the Problem. 
Thank you !

sn4k3, 2020-04-11:
Please strict follow the readme

karasu, 2020-04-20:
Hi,

To be able to select the profiles I had to change this line:
```
compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_CREALITY.*/ and printer_notes=~/.*PRINTER_MODEL_ENDER3.*/ and nozzle_diameter[0]==0.4 and printer_notes=~/.*PRINTER_NOT_VOLCANO.*/
```
to this:
```
compatible_printers_condition = printer_model=="ENDER3" and nozzle_diameter[0]==0.4
```
hope this helps.

sn4k3, 2020-04-20:
> PRINTER_VENDOR_CREALITY

Strange, can you confirm under Printer -> Notes if you have the following lines:

PRINTER_VENDOR_CREALITY
PRINTER_MODEL_ENDER3
PRINTER_NOT_VOLCANO

?

karasu, 2020-04-20:
Hi,
I have exactly this:
```
Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.
PRINTER_VENDOR_CREALITY
PRINTER_MODEL_ENDER3
```
I think the problem is that the new printers didn't show up after being copied (or I didn't see them, let me test it again as it's the first time I'm using PrusaSlicer).

sn4k3, 2020-04-20:
Theres one line missing, what profile you using?

karasu, 2020-04-20:
Creality ENDER-3

sn4k3, 2020-04-20:
Thats strange, it works for me... Maybe you installing profiles on wrong folder? Since i don't own linux i cant confirm, but works under windows

karasu, 2020-04-20:
No, no... it's the .PrusaSlicer folder, I'm sure. 

With the latest version 2.2.0 the Ender-3 printer profile does not have that "PRINTER_NOT_VOLCANO" by default.

Maybe you could update your README with something like "if the profiles do not show up, just add PRINTER_NOT_VOLCANO to printer->notes"...

Anyway, thanks for sharing your work!

sn4k3, 2020-04-20:
The Creality ENDER-3 printer is not my profile, is the original from prusa, so its normal that doesn't include the note. I will try see what's happening...

karasu, 2020-04-20:
I see. The problem is that after copying your files the new printers do not
show up.

I'm going to sleep now, but tomorrow I'll try it again with a clean
install, just in case I'm doing something wrong.

karasu, 2020-04-21:
I'm really sorry. I've tried again today in a clean installation and it's working now. Maybe I was doing something wrong and I didn't realize.

Thanks for your help!

sn4k3, 2020-04-21:
Seens fixed

@karasu 
"I'm really sorry. I've tried again today in a clean installation and it's working now. Maybe I was doing something wrong and I didn't realize.
Thanks for your help!"

closed 2020-04-21 by sn4k3