sn4k3/Ender3 #30: [QUESTION] Print quality in Overhangs

Astrofreak85, 2020-05-13:
With your profiles, when I reduce the max extrusion to 50mm/s I get good results, else my extruder sometimes skipped.
But I have a problem with overhangs, like the front of Benchy. The layers thee alway curl ob while being printed. When I slice Benchy with Cura, and stock settings it looks good... 
What is causing the problem?

Astrofreak85, 2020-05-13:
[image]

sn4k3, 2020-05-13:
Need more information, such as:
Is ender stock or modded?
Filament used
Temperature used
printer profile + print used

Astrofreak85, 2020-05-13:
Ender 3 pro
Not modded, just Metal Extruder an other cooling.
Filament, PLA from "Das Filament" 210°C
Profile: yours Ender 3 stock, Print speed is yours, not change for the print above

currently I am putting a print with same speed settings as used in Cura

When I compare Cura and Prusa print after the main deck of benchy is ready, it seems prusa uses much thicker walls...

[image]
[image]
[image]

Astrofreak85, 2020-05-13:
The additional fill between the two outher perimeters seems to corelate with number of upper an lower layers in PrusaSlicer, at least changing this has an effect here, wenn i reduce upper and lower to 3 ist gone, and the print looks better, but still not perfect
strange how different the slicers behave...
maybe this has nothing to do with your profiles at all...

sn4k3, 2020-05-13:
Diferent software have different planers and ways of print, it look like a overhang problem. You can try reduce overhang speed (bridge) and increase cooling fan speed on those cases which for PLA is always 100%.
Also make sure you use latest prusaslicer + my profiles
You can try set every setting equal to CURA and see if problem stays or not. I printed a bench two days ago with PLA and i don't have that problem, i'm using 0.65mm lines and 0.6mm nozzle which is worst for overhangs (more mass)

closed 2020-06-13 by sn4k3