sn4k3/Ender3 #32: [QUESTION] Why no 0.4 nozzle profile for Direct Drive?

MiniNinja1, 2020-05-30:
I'm curious why you don't have profiles for a direct drive with a 0.4 nozzle. It's a common nozzle size.
I already made one myself based on some of your settings and it's been printing fine.

sn4k3, 2020-05-30:
Most of profiles were made based on my usages or on requests.
Do a 0.4mm nozzle for DD is very easy, just copy stock profile and change retraction

MiniNinja1, 2020-05-30:
How about other nozzle sizes besides the one you've given? Do I simply change _Nozzle diameter_ and _Max/Min Layer height limits_?

sn4k3, 2020-05-30:
If i have that nozzle size configured and you just want convert Bowden to DD then Yes.
If not and let's say you want configure a 0.5mm nozzle, then no and you need to create each print profile with correct extrusion widths and calculate max speeds given the set extrusion

MiniNinja1, 2020-05-30:
Okay, I see what you mean. I think I'll stick with my 0.4 nozzle for now and do the necessary adjustments. Thanks!

closed 2020-06-13 by sn4k3

o5, 2020-10-30:
I've upgraded my Ender 3 PRO by this [guide](https://www.3dpedie.cz/en/inspirations/octoender_by_kiron). I used your profile for stock Ender before but now I'm looking for DD with 0.4mm Nozle and P.I.N.D.A. I started a new profile based on your DD 0.6mm Nozle (no Volcano). Any tips from you? :) 

I changed: Max print height, Commented out G-CODE for BLTouch, Nozzle diameter, Set Retraction length to 1.5mm and speed to 120mm/s. 

I can prepare PR if you are open to add a new profile.

sn4k3, 2020-10-30:
Direct drive with 0.4mm nozzle you can clone stock ender bltouch and change the the Retraction length and speed, everything will be the same, also adapt the start gcode for the probe that you are using.
Remove PRINTER_HAS_BOWDEN and add PRINTER_HAS_DIRECT_DRIVE on printer notes.
As addition if you are using linear advance, set a model ex: PRINTER_FULLMODEL_ENDER3_DD_04