sn4k3/Ender3 #33: [QUESTION] There is no stock 0.4 nozzle "print" profile

rufik, 2020-12-17:
Do I see right there is no "print settings" profile for stock 0.4 nozzle?

sn4k3, 2020-12-17:
There are, are you with any beta PrusaSlicer version?

rufik, 2020-12-18:
I'm using PrusaSlicer 2.2.0, so no bete here. When I go into `PrusaSlicer/print` folder there are print profiles for 0.40 nozzle like that:
[image]

Which one is for stock hotend? I can see only for MK3 Volcano ones...

pahowes, 2020-12-18:
If you're looking in the `PrusaSlicer/print` folder under your user profile, you'll see only those you added manually. The first number is the layer height, not the nozzle size. The second number `0.6 nozzle MK3 Volcano` would refer to an E3D Volcano hot-end with a 0.6 mm nozzle.

The factory profiles are found in the application directory as `profiles/PrusaResearch.ini`. They're all lumped together in that file.

sn4k3, 2020-12-18:
https://github.com/sn4k3/Ender3/tree/master/PrusaSlicer/print

Stock nozzle files ends with: Ender3.ini

closed 2021-04-21 by sn4k3