sn4k3/Ender3 #35: [BUG] Settings not working for PrusaSlicer 2.3

tb205gti, 2021-02-20:
**Describe the bug**
The settings does not show up when using PrusaSlicer 2.3

**To Reproduce**
Steps to reproduce the behavior:
install latest PrusaSlicer (2.3) - copy the profile files over - prusa shows the following message when starting Prusaslicer:

"You have the following presets with saved options for "Print Host upload":
    "Creality Ender-3 DirectDrive Volcano 0.6 Nozzle + BLTouch",
    "Creality Ender-3 DirectDrive Volcano 0.6 Nozzle + BLTouch RECALL",
    "Creality Ender-3 Volcano 0.8 Nozzle",
    "Creality Ender-3 Volcano 0.8 Nozzle + BLTouch",
    "Creality Ender-3 Volcano 0.8 Nozzle + BLTouch RECALL",
    "Original Prusa i3 MK3 0.6 nozzle Volcano",
    "Original Prusa i3 MK3 0.6 nozzle Volcano NO PINDA"

But since this version of PrusaSlicer we don't show this information in Printer Settings anymore.
Settings will be available in physical printers settings.

By default new Printer devices will be named as "Printer N" during its creation.
Note: This name can be changed later from the physical printers settings"

**Expected behavior**
Expect it to have the profiles as on 2.2

**Screenshots**

**Desktop (please complete the following information):**
 - OS: MacOS Catalina
 - Version 10.15.7

**Additional context**
Add any other context about the problem here.

sn4k3, 2021-02-20:
Yes, i have to fix them for new version, also i need to update the new fields that was introduced

nurglexp, 2021-04-17:
I get great results using your profiles in 2.2, really looking forward to new profiles for 2.3. Will you be releasing updated versions soon?

sn4k3, 2021-04-17:
> I get great results using your profiles in 2.2, really looking forward to new profiles for 2.3. Will you be releasing updated versions soon?

I can upgrade them yes, but now prusaslicer have official profiles for ender3 and others, did you try them?

nurglexp, 2021-04-17:
I didn't want to upgrade to 2.3 as I saw your profiles didn't work. I had such good prints using your profiles I stuck with 2.2 but there are some features of 2.3 that interest me. If you think the inbuilt profiles with 2.3 are good i'll give them a go.

sn4k3, 2021-04-17:
My profiles will work with 2.3, only the new fields will have some different default values. But yeah you can try the official ones, if you find them worst tell me back and i will make a push

nurglexp, 2021-04-21:
I've tried the inbuilt creality profiles in version 2.3.1
They seem ok, I'd still prefer to use your profiles if you have time to push an update.
Many thanks!

sn4k3, 2021-04-21:
You use the stock profile or other?

closed 2021-04-21 by sn4k3