sn4k3/Ender3 #36: [QUESTION] Direct Drive for 0.4 Nozzle profile

NovaViper, 2021-05-02:
Hey, I'm wondering if you could make a profile for 0.4 nozzles with direct drive

sn4k3, 2021-05-02:
It's easy to do:

1. Clone "Creality Ender-3 Stock" profile
2. Change retraction settings to:
[image]

NovaViper, 2021-05-02:
Ah cool. Thanks for the info! Uh... how do I go and get the filament profiles for the machine?

sn4k3, 2021-05-02:
You can get them from Official creality filaments on PrusaSlicer, you just need to go to PrusaSlicer Wizard and select Ender3 Printer and it filaments and i think it will show up

NovaViper, 2021-05-03:
Ah ok, thanks!

closed 2021-05-03 by NovaViper

Moxie1776, 2021-05-03:
Be careful with those settings, sometimes with length that hi it can cause your hotend to clog. Speed can be adjusted up too, I would play with those settings. You may want to try zlift starting at .2.

sn4k3, 2021-05-03:
0.8mm is not that high. My settings are for 3ed setup similar to prusa, but of course there are no universal setting, adjust if needed

Moxie1776, 2021-05-04:
> 0.8mm is not that high. My settings are for v6 e3d setup similar to prusa, but of course there are no universal setting, adjust if needed

If I go past .6 my all metal microswiss clogs...

NovaViper, 2021-05-05:
> > 0.8mm is not that high. My settings are for v6 e3d setup similar to prusa, but of course there are no universal setting, adjust if needed
> 
> If I go past .6 my all metal microswiss clogs...

I actually have a V6 and BMG (both are clones) setup with the BMG directly connected to the V6 and dual 5015 fans. The retraction length and speed is actually pretty normal for a V6, anything above 1mm is too much.