sn4k3/Ender3 #40: [QUESTION] I want to use BLTouch in the ender3V2 in Klipper , but i don't know the pin number.

2718564960, 2021-11-29:
I want to use BLTouch in the ender3V2 in Klipper , but i don't know the pin number.
This is the picture, thank you .
[image]

plazmax, 2021-11-29:
i think u asked wrong repo this is for prusa slicer ender 3 profiles, but u can check theese:

https://youtu.be/sWKLxxFEEbc?t=497
https://www.klipper3d.org/BLTouch.html

https://www.smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/
https://github.com/Klipper3d/klipper/blob/master/config/printer-creality-ender3-v2-2020.cfg
https://all3dp.com/2/klipper-ender-3-v2/#i-9-changes-for-a-bltouch-sensor

2718564960, 2021-11-29:
> i think u asked wrong repo this is for prusa slicer ender 3 profiles, but u can check theese:
>
> https://youtu.be/sWKLxxFEEbc?t=497 https://www.klipper3d.org/BLTouch.html [...]

sorry, thank you !

closed 2021-11-29 by 2718564960